nettigo/namf #3: Content-Length in POST request not filled

pkozak2, 2020-03-07:
My custom API (.NET Core) returns 411 HTTP Error.

I found that the Content-Length is not filled in request headers.

Current BETA NAMF-2020-23
[image]

Stable version - NAMF-2019-020:
[image]

Additionaly I found that used Library probably want to fill Content-Length:
[https://github.com/esp8266/Arduino/blob/372a3ec297dfe8501bed1ec4552244695b5e8ced/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp#L673](https://github.com/esp8266/Arduino/blob/372a3ec297dfe8501bed1ec4552244695b5e8ced/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp#L673)

But, not only I have this issue with this library:
[https://github.com/esp8266/Arduino/issues/2678](https://github.com/esp8266/Arduino/issues/2678)

pkozak2, 2020-03-07:
I tested airohr firmware which using HTTPClient like new nettigo firmware, and it's working with my API, so there is something wrong with new source.
I will try to find what, but I can't promise

irukard, 2020-03-08:
It's simple. Basic auth isn't working yet.

pkozak2, 2020-03-08:
Ok, I will try today.

netmaniac, 2020-03-08:
@irukard I'm not sure if this really BasicAuth issue. It works for me with Influx (so it is not a custom API though), in attached traffic auth headers seems to be exact the same. API responds with 411 not 401, so auth is working. 
I'll take a look at code I guess we are not adding content-length :)

netmaniac, 2020-03-08:
@pkozak2 I have added Content-Length header when sending. As I thought it was lost when converting code.Try with new FW (2020-24). If it won't work for You reopen issue.

closed 2020-03-08 by netmaniac

pkozak2, 2020-03-08:
As Ikurad says, authorization header not working for now.

#4 fixes this issue.

commit 7afa04dc (2020-03-08): add content-length to sendData. Should fix #3