nettigo/namf #8: [question] access to unstable builds

sredni, 2020-04-28:
Is there any possibility to access builds listed in Versions.md?

netmaniac, 2020-04-28:
I'm not sure what You are asking about? Historical versions? Now it is 2020-28, You are asking about previous builds?
I forgot to tag commits with versions, so there is not easy way to get commit hash. Probably You should annotate Versions.md and will see when each entry was updated. That commit should be proper to build previous version.
Unless You are asking about something different. Anyway, from now I will make tags on releases...

sredni, 2020-04-28:
Yes, that's what I'm asking. So tagging will be sufficient to achieve that, but I propose to use "release" Github feature. This will automatically create tag and allows to upload compiled bin files in case someone want a specific version.

netmaniac, 2020-04-28:
Yeah, I was thinking about releases.
Last few binaries (from version 2020-23) I have backed up, so in case You need some of them I can share.

sredni, 2020-04-29:
I already handled that via Versions.md history, thanks :)

closed 2020-04-29 by sredni