nettigo/namf #14: Git Workflow and rules info in Readme

Havelock-Vetinari, 2020-08-14:
It would be convenient to have information about pull request workflow in this project. Should they be requested to master or to beta branch? Should bugfixes and new features be submitted in the same way?

I'm implementing support for mqtt as api and for pir sensor for turning on and off lcd backlight, and I'll be happy to contribute them to main project in the most welcome way.

irukard, 2020-08-14:
Good point. The short answer is "beta".
The long answer: We need to think about workflow. We would appreciate any suggestions.

Havelock-Vetinari, 2020-08-14:
I have a good experience with github workflow with only master branch and feature branches. Releases are just tagged commits in master, other commits are snapshots/nightly builds. But backed up by automated tests of builds. As far as I know, it is possible with platform.io.