nettigo/namf #21: Display Winsen MH-Z14A CO2 Sensor readings on LCD

icnmfabro, 2020-11-28:
The sensor reading is currently only online on the http://aqi.eco portal. It would be interesting to present the data directly in the LCD display.

netmaniac, 2021-01-08:
Display for sensors in new scheduler is now supported. Now I'm moving Winsen support to new scheduler, then it will display results to LCD.

icnmfabro, 2021-01-08:
> Great ... I have already installed the beta version ... Greetings from
> Massimiliano.

Great ... I have already installed the beta version ... Greetings from
Massimiliano.

netmaniac, 2021-01-16:
NAMF-2020-35rc3 - new beta is on server. Winsen is supported by new scheduler. After update You have to enable sensor again, this time in bottom of config page. There should be also option to use LCD display.

icnmfabro, 2021-01-17:
Hello.

I have enabled the sensor and enabled the Winsen data display on the LCD, but there is a problem.
On the LCD I signal that there are 6 display pages but when I arrive at the 5th page it jumps back to the first data display screen (actually the winsen data are not displayed).
Can you check and confirm the problem?

I'm waiting for your new ones.

icnmfabro, 2021-01-18:
Other bug found. by clicking on the selection for displaying on LCD the SPS30 sensor, or by leaving it unchecked, the sensor data are visible on the LCD screen.

netmaniac, 2021-01-18:
> I have enabled the sensor and enabled the Winsen data display on the LCD, but there is a problem.
> On the LCD I signal that there are 6 display pages but when I arrive at the 5th page it jumps back to the first data display screen (actually the winsen data are not displayed).
Which one LCD Your NAM has? 16x2 or 20x4?

icnmfabro, 2021-01-19:
20x4 lcd

netmaniac, 2021-01-19:
There is NAMF-2020-35rc4 release. It should fix disabling LCDs for SPS30 and Winsen. 

I could not reproduce wrong behavior with Winsen and display. It shows reading correctly. Do sensor restart when gets to 6/6 or cycles through display and takes measurements and sends data to API?

Does sensor display CO2 reading on /values page?

icnmfabro, 2021-01-20:
After updating to release 35rc4 I deleted the namf configuration and entered again.
I found this:
1) Now if you remove the check to the "Display on LCD" box on the SPS30 sensor, actually screens 4 and 5 containing the sensor data disappear. so this problem seems solved.
2) For the Winsen CO2 sensor, if the SPS30 data display is removed, then the Winsen MHZ14A data screen 4 appears on the lcd display, if the SPS30 data display is also active (precisely screens 4 and 5), the Winsen data (now screen 6) is skipped and not displayed. Everything then restarts with the start of the display cycle from screen 1.
So in summary, if I activate the display of the SPS30 I cannot view the CO2 data on the display, otherwise if I delete the display of data on the SPS30 LCD then the CO2 screen appears.

The data at https://meteofabro.aqi.eco/en/12954058 seems to work.

netmaniac, 2021-01-20:
> 2. For the Winsen CO2 sensor, if the SPS30 data display is removed, then the Winsen MHZ14A data screen 4 appears on the lcd display, if the SPS30 data display is also active (precisely screens 4 and [...]

So problem is present when both SPS30 and MHZ14A are enabled and checked for display? I check such variant, since I had separate NAMs with SPS30 and MHZ. I need to place both sensors into one NAM and verify.

icnmfabro, 2021-01-20:
Yes exactly ... the problem occurs with both sensors enabled for display.

netmaniac, 2021-01-20:
NAMF-2020-35rc5 on update server. For me it fixes problem, please check.

icnmfabro, 2021-01-20:
Yes, the LCD display problem has been solved ...
Fantastic... :-)

icnmfabro, 2021-01-20:
I would also like to notify you of a data problem with the API of https://luftdaten.info/.
On my https://devices.sensor.community/ account I added the SPS30 sensor, but I can only see the data from the BME280.
Both sending to Luftdaten and Madavi are enabled in the NAM configuration.
Can you check if it's a firmware problem?
Thanks.
[image]
[image]

netmaniac, 2021-01-20:
Yeah, only old style sensors are supported when sending data to LD. Need to rethink how to implement that with new scheduler.

closed 2021-01-20 by netmaniac