nettigo/namf #23: SPS30 not sendng data to LuftDaten

netmaniac, 2021-01-20:
I would also like to notify you of a data problem with the API of https://luftdaten.info/.
On my https://devices.sensor.community/ account I added the SPS30 sensor, but I can only see the data from the BME280.
Both sending to Luftdaten and Madavi are enabled in the NAM configuration.
Can you check if it's a firmware problem?
Thanks.
[image]
[image]

_Originally posted by @icnmfabro in https://github.com/nettigo/namf/issues/21#issuecomment-763917579_

netmaniac, 2021-01-20:
In current firmware results from sensors managed by new scheduler are not being sent to LD. Plan is to fix in next release (NAMF-2020-36)

icnmfabro, 2021-01-21:
OK...

sumerland, 2021-02-16:
Does this also apply to the Sensirion SHT3x ? I am running version 2020-36rc1 and would like to see the results on luftdaten & madavi/grafana. Should I switch to the alpha build?

netmaniac, 2021-02-17:
On beta channel there is new release NAMF-2020-36rc2, with support for sending data to SensorCommunity from sensors managed by new scheduler.

To send data from SPS30, SDS011 has to be disabled. Do it **only on NAM without SDS011 mounted**. With SDS011 disabled in config and sensor being present on board, SDS011 will run in continuous mode and laser diod will use it's life time quickly.

@sumerland SHT3x should be sending data to SC in 36rc2, but I had no sensor with SHT3x to test it, so support was compiled without test on physical sensor. Let me know if 36rc2 sends SHT3x  data to Sensor Community. If there will be issues I will connect find some spare SHT30 to connect it to NAM and test...

netmaniac, 2021-02-17:
I have removed that condition for SPS, so now my dev sensor sends both SDS and SPS to SC. Looks like it results in storing readings from SPS30 in SC every time (NAMF first will send data from SDS and later from SPS so maybe that is the reason)

Anyway - probably will do rc3 without that check, so each sensor with SDS and SPS will report data from SPS.

sumerland, 2021-02-18:
@netmaniac Thanks for the update! NAM is running with rc2 for 6h now but I still don't see SHT35 data neither on maps.sensor.community nor api-rrd.madavi.de (other sensors' data is there). The SHT35 was enabled in devices.sensors.community 2 days ago.

icnmfabro, 2021-02-18:
Hello.

The SPS30 sensor data finally appeared on the Luftdaten.info community.
Yeahhh
[image]

sumerland, 2021-02-18:
@icnmfabro Interesting to see that also on your SPS30 the PM10 is almost identical to PM2.5. It's the same over here.

@netmaniac Since tonight's update the SDS11 data is missing quite often. It's just the PM data, the other sensors are doing fine:
[image]

[image]

icnmfabro, 2021-02-18:
> Since tonight's update the SDS11 data is missing quite often. It's just the PM data, the other sensors are doing fine:

It was found that the SPS30 sensor has similar PM2.5 and 10 readings, compared to other sensors.
It would be nice to know if this is the most accurate sensor or are the others.
The only sure thing is that the purchase cost of the SPS30 is certainly higher than the others.
We hope for the goodness of the project and the build quality of this SPS30.

sumerland, 2021-02-20:
@netmaniac Thanks a lot for rc3 - SHT3x data is visible on maps.sensor.community and api-rrd.madavi.de now.