nettigo/namf #24: Build instructions

dblachut, 2021-03-19:
Would you mind sharing instructions how to build/config depending on NAM version?

I haven't found anything yet and I would like to be sure I am doing everything correctly.

More specifically I would like to build the firmware uploaded into NAM 0.3.3 STD with some modifications and possibly upload some PR.

If I am not mistaken the platform should be `espressif8266` and the board is `d1_mini`?

netmaniac, 2021-03-22:
It looks like something changed in SSD1306 library definition. Older versions are not available via Platformio, so installing SSD1306.h  fails. I have update `beta` branch with library bumped to 4.2.0. I have not tested it with OLED display, but at least whole project compiles.

So - checkout project, switch to beta branch and `pio run -e nodemcuv2` should build default firmware (Polish language). To get English version use `nodemcuv2_en`

More detailed info how to contribute - in preparation.

netmaniac, 2021-03-22:
With current code there is no difference in firmware between STD/PRO versions.

dblachut, 2021-03-22:
Great, I was more concerned about platformio parameters, especially board `nodemcuv2` does not seem to be correct.

netmaniac, 2021-03-22:
Why not? https://docs.platformio.org/en/latest/boards/espressif8266/nodemcuv2.html To be honest it is inherited from Ludtdaten firmware, there is https://docs.platformio.org/en/latest/boards/espressif8266/d1_mini.html however I don't see practical differences between them:

https://raw.githubusercontent.com/platformio/platform-espressif8266/master/boards/d1_mini.json
https://raw.githubusercontent.com/platformio/platform-espressif8266/master/boards/nodemcuv2.json

dblachut, 2021-03-22:
I didn't investigate it deeply, I just saw that there is a separate configuration for Wemos D1 mini and it raised my suspicion. :)

Thanks for the detailed explanation I will close the issue once I will build and upload the firmware successfully.

dblachut, 2021-03-22:
Unfortunately after applying the [change ](https://github.com/nettigo/namf/commit/6358dd886d2260a09f170e5440565e01cec19c06) manually I am still having an error:

> Library Manager: Installing ESP8266_SSD1306 @ 4.2.0
> Warning! Could not find the package with 'ESP8266_SSD1306 @ 4.2.0' requirements for your system 'windows_amd64'

I didn't find a library under this name (ESP8266_SSD1306) in platformio.org search. The only ones that are found with such header are here: https://platformio.org/lib/search?query=header%253ASSD1306.h%2520platform%253Aespressif8266%2520framework%253Aarduino&page=1

Anyway I am able to comment out the code related to SSD1306 and build the project.

netmaniac, 2021-03-22:
Library name was also updated in current beta.

dblachut, 2021-03-29:
Sorry for a delay but I can now confirm that it works! Thanks a lot. I am also able to update the firmware remotely using /forceUpdate page. 🍾

dblachut, 2021-03-29:
Actually I have created some wiki page with build & upload instructions. Please take a look: https://github.com/dblachut/namf/wiki/Build-&-upload-firmware

dblachut, 2021-04-07:
@netmaniac did you get any chance to take a look at my wiki page proposal?

netmaniac, 2021-04-10:
I have missed notification. Yes, seems OK. There is OTA option - visit /ota and for 60 seconds You can upload code. Also (feature still in beta line, but will be in next stable release) You can set additional upload options (meant is for upload, but You can override anu platformio.ini option) in upload_options.ini. Example for OTA: https://github.com/nettigo/namf/blob/beta/upload_config.ini.example This file is in .gitignore so sensitive information won't be added to repository.

netmaniac, 2021-04-10:
Also - Platformio is something much bigger than VCS extension :)

dblachut, 2021-04-10:
Alright I will test the OTA option and try to describe it on the wiki page as well.

Yeah I guess that it is but the IDE is not a separate program but an extension to VSC.
Do you want me to change this info somehow?