nettigo/namf #25: Support for airmonitor.pl data sending

dblachut, 2021-03-24:
It would be nice to have support for airmonitor.pl data sending next to other services.

I will try to implement it unless author has something against it.

netmaniac, 2021-03-25:
I don't think it is necessary, but feel free to question my beliefs :) 

NAM sends data to SensorCommunity/Luftdaten. Airmonitor.pl does import data from Luftdaten, thus data from NAM is available in Airmonitor. There will be hard to avoid duplication of data if NAM will send data through Luftdaten and airmonitor.

dblachut, 2021-03-25:
Actually I was wondering if they don't import the data from SensorCommunity however I couldn't find any official statement that they do. Moreover I didn't saw my sensor on the map but I guess it may be visible after some delay.

Thanks for explanation.

closed 2021-03-25 by dblachut