nettigo/namf #27: Stability issue with NAMF-2020-37

komar007, 2021-04-29:
There seems to be a stability issue after automatic upgrade to NAMF-2020-37.
The screenshot from madavi shows some random blanks in 7-day data that start on April 28.
[image]
PM2.5 data for April 27:
[image]
And PM2.5 data for April 28:
[image]
This seems to correlate with NAMF-2020-37 release date...
I have been monitoring uptime for the last hour, and it seems that watchdog reboots are rather frequent, I haven't seen an uptime longer than 30 minutes.

What debug data can I provide?

netmaniac, 2021-04-29:
Hi!
So, reboots are inducted by watchdog? On /values page is section NTW WTD. Are there 'status' values other than 0?

--

komar007, 2021-04-29:
@netmaniac I cannot see "NTW WTD" on /values page, although the last reset reason is exception, see screenshot:
[image]

netmaniac, 2021-04-29:
Isn't that /status page not /values?

netmaniac, 2021-04-29:
But - reset reason - exception, so it is not watchdog

Does page /stack_dump offers something more than date and FW version? If yes, could You paste it?

komar007, 2021-04-29:
Yes, that was the status page. Values page is like this:
[image]
I can't find "NTW WTD" here.
As for resets - the last reason was exception, some earlier one was watchdog.
I started collecting uptimes until reset now, so far: 23m, 39m. Currently running for 13m, last reset reason: Exception.

netmaniac, 2021-04-29:
Could You re-enable watchdog? Save wtd config again - will it make NTW WTD section to show on /values page? 

And could You write an email to info@nettigo.pl - if You don't use Influx, then Your sensor may send data to our internal Influx server, and we will have more insights what is going with sensor. Dont see  way to send login data here over githhub....

komar007, 2021-04-29:
I have enabled NTW WTD and it appeared on /values page. I have configured it to ping my router, but I have never used this feature before, so it can't have been the reset reason.
[image]

sumerland, 2021-04-29:
I am still running 37-rc5 (sensor did not update to 37) and saw missing data for fine dust but it recovered on its own without a reboot (14d uptime right now). It did not affect the attached SHT35 and BME280 - these are pretty much complete:[image][image]

komar007, 2021-05-02:
@sumerland, It looks like you've had some interesting problem, but likely a different one than mine. In my case lack of reported data has always concerned all the values and has always been related to frequent reboots (no more than 2h uptime). Additionally, I have bisected the recent changes in the repository and found that in case of my device the problem exists since at least 37-rc2, while 36 works fine. If that wasn't enough, @netmaniac has not managed to reproduce the problem elsewhere, so it seems it's somehow related to just one device (that we know of).

I'll be posting full results of my investigation shortly.

komar007, 2021-05-02:
Bear with me and my debugging methods, because I've been too lazy to unmount and take home my sensor to collect logs.

I have bisected the master branch to find that the culprit is 5e207a887d0b3e1c3d71c4ab84b0048fed80573d.
Reverting this commit (against NAMF-2020-37) heals my problem completely.

Next, I have tested various combinations of places where I reverted the change from RAM to PROGMEM for string literals. Long story short, a single change of a string literal `"/config.json"` to `F("/config.json")`  in https://github.com/nettigo/namf/commit/5e207a887d0b3e1c3d71c4ab84b0048fed80573d#diff-9f24668b43bfb933b02221fb9f6dc1a442783d6c6ac99a010074313f03e1311fL924 caused my frequent reboots and missing data reports. Reverting this single-line change against faulty NAMF-2020-37 fixed the issue for me.

This was the what, now about the why.

My current hypothesis is that we're simply dealing with a stack overflow here (which might happen only in my case for some specific configuration reason, or maybe network conditions, order of function calls, memory allocations, etc. I currently reject a possibility that this has to do with incorrect PROGMEM treatment (32-bit unaligned reads), because I have not found any evidence for this in the code.

According to my quick glance at https://github.com/esp8266/Arduino/blob/master/cores/esp8266/WString.h and https://github.com/esp8266/Arduino/blob/master/cores/esp8266/FS.h, moving the strings to PROGMEM may actually increase memory consumption. The method `remove` to which the strings are passed is overloaded for `const char*` and `const String&`, which means that in case of `F("/config.json")`, first a temporary `String` object is created using its converting constructor from `const __FlashStringHelper*` to be later passed by const reference to `remove`. The `String` implementation has a fixed overhead of 12 bytes (11 of which can be used as data buffer), and since the string used in this case is 13 bytes long, it causes an extra heap allocation on top of those 12 bytes on stack. If I understood the code correctly, `String` copies the data to RAM on construction, so the only advantage of using `F()` is trading static memory in RAM for dynamically allocated memory on stack / heap depending on the situation. Either this extra allocation or just stack footprint of String seems to kill my NAM.

I confirmed the above by replacing the string literal with a temporary `String(literal)`, as in:
```if (SPIFFS.remove(String("/config.json"))) {```
to rule out using PROGMEM as the root cause, and got the NAM to reboot frequently, just as in case of `F()`, while using just string literal fixes the issue.

And now the best part is, I have no idea why `webserver_removeConfig` should ever be called, because I never use this endpoint and I checked in the code, that only the webserver calls this function.

komar007, 2021-05-04:
I have tried removing the config file (is that the only way to factory reset the device?), but that did not help on 2020-37 (without my patch).
Another observation is that the web interface in inaccessible for a significantly longer time during measurement on the new version. For 2020-36 it is about 10 seconds, whereas for 2020-37 at least 60 seconds.
I think I'll stick with 36 for now. Hopefully this will be reproduced somewhere else, so that there is more data to work with...

netmaniac, 2021-05-09:
> According to my quick glance at https://github.com/esp8266/Arduino/blob/master/cores/esp8266/WString.h and https://github.com/esp8266/Arduino/blob/master/cores/esp8266/FS.h, moving the strings to [...]

Idea for moving string to PROGMEM is that saves RAM globally. Without storing strings in PROGMEM all strings are copied to RAM before binary is being run. So, every single "plain" string literal in code uses RAM all the time. String wrapper creating new variable to store string in RAM exists only when give variable is used. So, in total You have more RAM to use, since all strings from code "not being run currently" are not eating memory.

This is strange if that change have fixed uptime for Your sensor, since SPIFFS calls are only on startup (readConfig), saving configuration or deleting. However, since not all calls to SPIFFS can be overloaded with __FlashStringHelper (SPIFFS.open is exception), so using F in SPIFFS calls does not make sense, since always one string (used with `open`)  will remain "plain" and will consume RAM.

So, in next beta (38rc1) F() macros in SPIFFS calls will be removed

komar007, 2021-05-09:
> This is strange if that change have fixed uptime for Your sensor

Yes, it's strange. In fact it's rather not moving anything back from PROGMEM that fixed my problem. The problem is likely more complicated, but this is just how it manifests itself. I have given up debugging though.

> so using F in SPIFFS calls does not make sense, since always one string (used with open) will remain "plain" and will consume RAM.

True. I noticed that but actually forgot to mention.