nettigo/namf #30: 2020-38rc1 no longer displays the SDS011 data with the display 2004.

icnmfabro, 2021-05-13:
Hello.

I encountered a problem after the 38rc1 upgrade.
After the update the LCD 2004 no longer displays the SDS011 sensor data page.
Can you check?

netmaniac, 2021-05-15:
New SDS011 and HECA support have separate setting for enable sensor and enable display. It can be reenabled on config page.

In next release (38rc4) old config setting for SDS/HECA will enable both sensor and display, so no change in config will be needed.

**For both SDS and HECA now are two config options** Old ones are still available. If SDS or HECA are enabled in old config section, then, no mater what is set in new config section, at start they will be enabled. With release 38rc4 the same behavior will work for display. If SDS/HECA is enabled in old section, both sensor and display will be enabled on start.

netmaniac, 2021-05-16:
NAMF-2020-38rc4 on beta update channel, should fix that.

closed 2021-05-16 by netmaniac

icnmfabro, 2021-05-16:
👍🖖