nettigo/namf #31: [Feature request] Send device ID via mDNS

bieniu, 2021-05-17:
Home Assistant uses `zeroconf` to discover a NAM device on the local network and propose a configuration of the integration to the user. The hostname is checked, if it contains `NAM-` that means this is a NAM device.

If the user changes the **network name** in the NAM device configuration, the hostname of the device will also change and the HA will not be able to discover the device.

Adding an additional text `ID` information with the default hostname (`NAM-12345678`) would solve this problem.

This is an example of mDNS information from Shelly device:

[image]

netmaniac, 2021-05-17:
It should be supported in 38rc6. Code is in beta, so if You want You may test it by building current beta and uploading to sensor. Or wait for next beta release to check. But for me it should work :smiley: 
[image]

bieniu, 2021-05-17:
Great, I'm using beta to test new features.

netmaniac, 2021-05-20:
New beta on update server (38rc6)

closed 2021-05-20 by netmaniac

netmaniac, 2021-05-21:
Can be done. So, new TXT field with key `manufacturer`, right? Also downcase on ID value is possible. Probably with next beta.

bieniu, 2021-05-22:
Great, thanks!

netmaniac, 2021-05-22:
OK, commit 8269356046547bb5f6c296fc0d2ca2287b4063f9 done that. With next beta will be deployed