nettigo/namf #33: (Question) Is HECA duty cycle working properly?

sumerland, 2021-05-25:
Running in 38rc6 now I have never seen HECA duty cycle values other than "0.0". I am sure HECA is active, as HECA T and RH are higher/lower than the other sensor readings. Is HECA data temporary (i.e. only showing a value different from 0 when active) or is it accumulated (how much % activity of sensor uptime)?

netmaniac, 2021-05-25:
I have created Details.md (in beta branch), there I will try do write a bit more about internals, how NAMF works and how to contribute to code. For now there is a little info about Duty Cycle: 

On /status page HECA reports duty cycle. This is how long alert was enabled due to humidity (DutyCycleRH) or temperature (DutyCycleTemp). DutyCycle is total time when alarm was active, no matter what condition has triggered it. DC is in %, so 50% means that alarm (and heater) was on during half time since last sending data.

Duty Cycle shown on /status page, and on LCD is from current cycle (so, it will be changing if heater is enabled). Values presented in /data.json are from previous cycle (sent to APIs).

netmaniac, 2021-05-25:
In other words - DC different than 0% with default SDS settings will be when HECA reports RH > 62%. Here is example from our Grafana.
[image]

sumerland, 2021-05-26:
Thanks for the explanation! That means, when RH > 62% the alarm is triggered, heater becomes which leads to HECA temperature > BME280/SHT3x temperature and HECA humidity < BME/SHT humidity. What puzzles me is that this is quite often the case but until now /status DC values have always been 0.00%. I will send two screenshots later tonight, when conditions are appropriate.

netmaniac, 2021-05-26:
Notice how HECA temperature/RH chart look - when heater is working then You get 'saw'. HECA readings are not shown on madavi. Also - BME is in separate case, to be not impacted by heater. 
To see that on charts, You need to use some Influx/Grafana or use other API. I can recommend You [aqi.eco](https://aqi.eco/en) and configure Your sensor to send data there. Then on graphs page select 'Detector humidity' and You will se RH reported by HECA. Use_Instantaneous_ to avoid graph smoothing by averaging. Example sensor: https://nettigo.aqi.eco/en/sensor-3/graphs

sumerland, 2021-05-26:
Apologies, I am bit slow today... I am just talking about the data shown on the NAMF webserver page (status and values) and the information visible on the attached LCD.  Duty cycle values are always 0.00, which indicates that HECA is not active. However, I can see that once ambient humidity goes beyond 62% (and/or iirc temperature drops below 5degC), then HECA temperature is higher than ambient T and HECA humidity is lower than ambient humidity. From the HECA sensor data I conclude that the heater is working and would expect duty cycle values between 0.0 and 100%.

I have come across the aqi.eco site before but I haven't figured out yet how to configure a sensor.

Here are two recent screenshots. I think it indicates that HECA is active, but the status page says 0.0%

[image]

[image]

sumerland, 2021-05-26:
Please forget everything I have said! I have just seen this

[image]

Question is answered - thanks for your patience.

netmaniac, 2021-05-27:
On LCD and /status page values of DC are from current measurement period.

closed 2021-05-27 by netmaniac