nettigo/namf #34: New feature: triggerActions/refreshReadings (after etering the page/clicking a button)

dblachut, 2021-06-01:
Sometimes you may suspect that the last reading was disturbed by some random event (e.g. it it much higher or lower than previous one).
In such case it would be great to have an opportunity to refresh the reading without waiting for the next one e.g. after entering the `/refreshReadings` sub-page.

**The main questions are**: is this something doable, will it get accepted as a PR, does it make sense?

If so, could you please provide some hints how to achieve this?

***
Next things that would be nice to have:
1. An option to have configurable physical button that would act as a trigger. (e.g. the used pin could be configurable)
2. Even more configuration that allows other actions to be performed by the trigger e.g.:
   * readings would be displayed on the LCD only after the trigger

Please let me know what you think!

dblachut, 2021-06-10:
@netmaniac did you get any chance to take a look and think about this feature?

netmaniac, 2021-06-15:
@dblachut 
There is problem with forced reading. For example SDS011 - it requires 30 seconds of warmup before taking some meaningful readings.  30 s according to SDS011 datasheet, SensorCommunity says after their tests, that 15 seconds is enough. That makes hard to take measurement on request. Also with other sensors, even those w/o mechanical parts - it should be rather some average/median from few measurements. 
Firmware now how long is measurement period so it may decide when is time to collect data. Having 'take measurement' call would be hard to achieve.

And regarding button - we have seen touch module from Heltec https://heltec.org/project/touch-button/ and from first tests it seems to be working through NAM cover, so maybe we will use such "button".