nettigo/namf #35: Status and future of BME280 as of 38-rc16

sumerland, 2021-08-04:
I read the release notes of rc16 but do not fully understand. Does that mean that support of BME280 will be removed from the firmware? The data is still showing up on the LCD and the website, but what is sent to api-rrd.madavi is obviously broken (maps.sensor.community seems to be fine).

Please keep support for the BME280.

[image]

[image]

[image]

netmaniac, 2021-08-06:
Sorry for confusion. 

BME will be supported, but due to lack of modules on the market we had to change our kits. New kits will include SHT31 for temp & humidity and BMP180 for pressure. So we have added support for BMP180 to fw. But due to my mistake data is being sent even when no BMP180 is enabled.

I have just released rc17 with fixed that problem with BMP and also BME is moved into new scheduler.

For BME difference is that temp/press/humidity are 5 times measured during measurement period and average is being sent not just result of single measurement before sending data.

netmaniac, 2021-08-06:
> . But due to my mistake data is being sent even when no BMP180 is enabled.

And that means that sent values are -128 for temp and -1 for pressure. That why graphs are distorted. With rc17 it should return to normal.

sumerland, 2021-08-06:
Thanks for the clarification and the fix! I was a bit worried that I had to scrap my BME280 ;-)

closed 2021-08-06 by netmaniac