nettigo/namf #36: PMS5003 gives no reading after NAMF-2020-38

blassepl, 2021-08-22:
After upgrading to NAMF-2020-38 PMS5003 stopped working. 

[image]

After connecting via serial no error can be seen in console, however PMSx003 reading are not providing any results. Tested on two devices, same result on both.

netmaniac, 2021-08-23:
I don't have PMS sensor available to test it, but could You switch one of sensor to beta update channel? I have released new beta with a bit bigger buffer size for PMS/SDS replies. This is only changed thing (regarding PMS) between 37 and 38. Does it work on new beta?

blassepl, 2021-08-24:
Still not working. Reading on new beta:
[image]

And the same on NAMF-2020-37:
[image]

netmaniac, 2021-08-24:
OK, I'll get one PMS sensor and will test it.

blassepl, 2021-09-06:
Sorry for the delay. Confirmed working with 40rc2.

closed 2021-09-08 by blassepl