nettigo/namf #37: SDS011 not refreshing after upgrade to NAMF-2020-38

pmendla, 2021-08-24:
SDS011 gets only first measurement after boot, then same value is constantly displayed on values page and uploaded to server. Other sensors (BME280, HECA) are refreshing as configured.

SDS011 works properly again after downgrade to 2020-37.

netmaniac, 2021-08-25:
Could You post all SDS011 related values from /status page (after few measurements, not just after start)?

Wojciech-X, 2021-08-25:
the same with my NAM, with new firmware SDS011 values are not being updated, only once just after reboot.
 
after about 2h from last reboot, on /status page in SDS011 section I have:

SDS011 | Status | 11
SDS011 | Status change | 6444528
SDS011 | Version data | 2018-11-16(07AC)
SDS011 | Liczba braków/wszystkich odczytów z SDS (zerowana przy każdej próbie aktualizacji) | 0/2
SDS011 | Checksum failures | 0.00% 0/33

pmendla, 2021-08-28:
NAMF-2020-40rc2 works well.

Can we expect NAM-2020-38 hotfix or we have to wait until official NAMF-2020-40 is released to switch back to stable updates channel?

netmaniac, 2021-09-08:
I'm sorry for delay @pmendla I will do stable release -40 in few days

netmaniac, 2021-09-09:
Release -40 is on update server.

closed 2021-09-09 by netmaniac