nettigo/namf #44: NTP server

awbnet, 2021-11-21:
@netmaniac Currently NAMF tries to fetch NTP server from DHCP server from WiFi, but not every DHCP server on home router has this set.

Can we have separate text field in NAMF, where we could enter our own desired NTP server? This could work like this:
* if NTP server field is empty then try to set time through DHCP fetched NTP server
* if NTP text field is populated with hostname then try to fetch time from this field

netmaniac, 2021-11-25:
AFAIK NTP is taken from NTP_SERVER which depends on language: https://github.com/nettigo/namf/blob/48e6e9e479b1b921b62510a804cd3c03313ae907/src/lang/select_lang.h so it should be enough...

awbnet, 2021-11-26:
Oh, so it is not taken from DHCP server. I didn't know that. Thanks for clarification.