nettigo/namf #45: Problem with version software 42rc6 and sensor winsen mhz14a.

icnmfabro, 2021-12-08:
Greetings.

I noticed that after the NAMF-2020-42rc6 update (2021-12-07 rev) my WINSEN MHZ14A - CO2 sensor no longer returns the values and from the LEDs you can see that it no longer takes CO2 readings.
Why is this sensor disabled when doing beta software updates?
Can you check and reactivate it? Thanks.

netmaniac, 2021-12-08:
I have one MHZ14A connected to my dev sensor and it is sending data all the time: https://nettigo.aqi.eco/pl/sensor-2/graphs 

Can You verify that connection is OK? And most common remedy in IT world :smile: - have You tried to power off sensor for few minutes and see if the  readings from MHZ will reappear?

icnmfabro, 2021-12-09:
Now it works, switched off and on again, or because it has been upgraded to version 42rc7.

netmaniac, 2021-12-09:
Changes between rc6 and rc7 was so small that I don't think this is the reason, rather it was restart.

closed 2021-12-09 by netmaniac