nettigo/namf #51: Heating element operation with smoother temperature and humidy

ProBackup-nl, 2021-12-21:
Heating element (HECA) seems to have a hysteresis of almost 5ºC:
[image]

and 18-19% variation for relative humidity:
[image]

It would be nice of these wavy lines for temperature and RH become smoother. Not by changing the graph but instead by changing the operation of the Heating Element Control Assembly (HECA).

netmaniac, 2021-12-29:
In future release (maybe it will be in -43) I plan to add config options for setting humidity and temp values when HECA start heating. Then can be hysteresis being more controllable. But current default values are ok in most cases since they keep air under 70% of RH and temperature over 0°C if it is really cold. And these are conditions for proper work of SDS011. Under -10°C and over 70% readings are unreliable.

jckkk, 2022-01-15:
I'm not an expert, but you should probably take into account the fact that you are heating the sensor. Therefore the readings cannot be directly read as air parameters. Just like when you expose a sensor to the sun, it won't give correct parameters of the surrounding air (will show higher temperature and lower humidity).