nettigo/namf #52: BMP180 pressure is too high

ProBackup-nl, 2021-12-30:
Nettigo Pro hardware with a BMP180 turns out to be having too high pressure readings.

Another BMP280 at 0.1 kilometer distance is: 1019.6
The Nettigo Pro hardware met BMP180 is at: 1061.2
Other pressure sensors nearby on maps.sensor.community show: 1019.6, 1019.3, 1019.3, 1018.0

netmaniac, 2021-12-31:
Have You replaced BMP to other one? It looks like faulty BMP for me. We have several devices in our location both with BMP180 and BME280 and don't see so high difference.

ProBackup-nl, 2021-12-31:
I didn't try another BMPX80 because I have only got a spare BMP280 sensor PCB thus missing "NAM I2C cable XH2.54-4P 10cm 24AWG" and "small sensor house" (the current one is filled with hot melt glue) .

Is there any warranty one the Nettigo Air Monitor (SKU: KIT-2159) that was ordered 13.12.2021?