nettigo/namf #57: Wifi stops when SDS fans stops

Kreid4, 2022-06-04:
Thank you for you work on this project.

We have just built and installed a NAM (NAMF-2020-43) KIT-2106  Nettigo Air Monitor (KIT 0.3.3 STD EN) - smog sensor, DIY Kit.
We have noticed that during a sampling cycle the LED flashes blue on the wifi chip for ~ 145 seconds. 
 However after the fan stops, the Blue LED stops flashing for ~60 seconds and we can not connect to the NAM via wifi. 
The overall cycling time is ~ 305 seconds.  The wifi signal when connected is good. 
Is this a normal sampling cycle for the device? 
We have also noticed that there are gaps in the data recorded and displayed in sensor graphs on Madavi.  See attached file.
Any advice would be very welcome please.
Thank you, 
Regards.
K.
Please note we also have a couple of sensor community kits and we have not noticed any wifi loss during  a sampling cycle.
[data-esp8266-5945275-2022-06-04.csv](https://github.com/nettigo/namf/files/8837976/data-esp8266-5945275-2022-06-04.csv)

Kreid4, 2022-06-07:
**Update**
We have enabled the HTTPS options for the sensor.community and madavi API uploads in the configuration settings.
The cycle is now 145 and 5s which is the same as the sensor community version of the device.  
So all is working normally now.  
However I would like to know how to view the debug data on the NAM /debug html page.  This is available as a scrolling text  at sensor_ip_addr/debug html page, with the sensor community firmware but not with the NAMF firmware.
Thank you

netmaniac, 2022-07-24:
Some basic functionality like in SC firmware (regarding log browsing) is available in -44rcX (beta line). So it will should be when -44 get released (I hope in few days).