nettigo/namf #59: SDS011 | Checksum failures | 16.92% 908/5365

nonitex, 2022-09-15:
Recently built a 0.3.3 STD kit from nettigo.pl, and the SDS has some checksum failures.
Firmware version: NAMF-2020-44rc4
Tried a couple of older firmware versions but the issue persists.

The logs I see from other people show 0% failures.

What are the possible reasons?
Is there anything I can do?

```
I2C | On I2C bus | 44, 76,
 
HECA | DutyCycle | 0.00 %
HECA | DutyCycleTemp | 0.00 %
HECA | DutyCycleRH | 0.00 %
 
SDS011 | Status | 3
SDS011 | Status change | 7853
SDS011 | Version data | 2021-5-16(6AD9)
SDS011 | Readings failed/total (counter reset with update check) | 2/97
SDS011 | Checksum failures | 16.92% 908/5365
 
WiFi | Time from NTP | Thu Sep 15 07:27:50 2022
WiFi | signal strength | -64 dBm
WiFi | signal quality | 72 %
 
NAM | Number of measurements: | 285
NAM | Uptime | 12h 17m 38s
NAM | Time from last update attempt | 12m 25s
 
NAMF | Sensor slots | 10
NAMF | Free slots | 7
NAMF | Sensors | SPS30 NTW_WTD SHT3x
 
ESP | Reset Reason | Software/System restart
ESP | Max Free Block Size | 23544/23544 B
ESP | Heap Fragmentation | 10/14 %
ESP | Free Cont Stack | 2144/2144 B
ESP | Free Memory | 26096/27456 B
ESP | Flash ID | 1458270
ESP | Flash Vendor ID | 94
ESP | Flash Speed | 40.00 MHz
ESP | Flash Mode | 2
 
ENV | Core version | 2_7_4
ENV | SDK version | 2.2.2-dev(38a443e)
 
DEBUG | Debug options
```

netmaniac, 2022-09-19:
It looks like current beta line has a higher checksum failure rate. Since, as result some readings are failed (no good packet from SDS) You can:

- switch back to stable firmware - from what I see NAMF-2020-43 should have lower chks failure
- make a bit longer time for collect readings - read time default 5000 ms means 4 packets from SDS. So If You make it a bit longer (10s) there will be lower chance for hitting all packets with CRC error

netmaniac, 2023-01-08:
Yesterday was new release NAMF-2020-45 - in our tests on beta line it has lower checksum error rates than -44 but still not so good as versions -43 and before. We will investigate that further, first we are waiting for more data from sensors on -45 if there was really improvement.