nettigo/namf #60: Air pressure history not available if BMP180/280 is set to be inside case

Leberer, 2022-09-17:
I noticed that the pressure seems not to be logged when the checkbox that the BMP180/280 is inside the case is checked. 
The pressure is still displayed for example on the sensor.community map, but no history is available.

When the checkbox is not selected, the pressure is indeed logged.

Also, I think the menu entry for this sensor is labelled incorrectly. It says "BMP180/280 humidity/temperature sensor" but isn't this family a temperature/pressure sensor instead?

netmaniac, 2022-09-19:
Sensor community ignores readings from BMP180/280 if just pressure is present. If You visit sensor page on madavi (assuming You are sending data there too) You will see both pressure from BMP and temp/humidity from SHT31. 
I will raise this issue with SC again, but I don't know if they change API behavior. 
Regarding description - You are right, I'll make change.

Leberer, 2022-09-19:
Interesting!
However, when I check madavi, it's the same behavior.
[This is the sensor ](https://api-rrd.madavi.de/grafana/d/q87EBfWGk/temperature-humidity-pressure?var-chipID=esp8266-11896695&var-type=BMP180&var-query0=sensors&orgId=1) and you can see when I disabled the checkbox on Sept 17th. No data available before (except from about January before I had enabled that checkbox).

netmaniac, 2022-09-19:
OK, I'll test that.

netmaniac, 2022-09-19:
I need to collect data a bit longer, but first few minutes with checkbox shows that both madavi like SensorCommunity API ignores readings when no temperature. But pressure is present in data. Influx shows readings also aqi.eco: https://nettigo.aqi.eco/en/sensor-2/graphs (check only last day - sensor had damaged BMP280 so readings are crazy).
SC does not show pressure data, only temperature/humidity from SHT31: https://maps.sensor.community/#15/52.2850/20.8890 sensor #75093.
I will report that to SC team again, but I don't know if they change API behavior...

netmaniac, 2022-09-20:
@Leberer  What is shown on /status page in BMx80 section? Which sensor type? 0 or 1? And could You post here content of /data.json from sensor?

Leberer, 2022-09-24:
@netmaniac I don't think there is a BMx80 section on the status page
This is the content on the status page.
<html><body>
<br>

 
--
I2C | On I2C bus | 44, 45, 77,
 
HECA | DutyCycle | 0.00 %
HECA | DutyCycleTemp | 0.00 %
HECA | DutyCycleRH | 0.00 %
 
SDS011 | Status | 7
SDS011 | Status change | 10055
SDS011 | Version data | 2021-1-12(B869)
SDS011 | Readings failed/total (counter reset with update check) | 0/127
SDS011 | Checksum failures | 2.76% 2010/72907
 
NTW WTD | Status code | 0
NTW WTD | Last check | 398 sec
NTW WTD | Pings | 10
NTW WTD | Responses | 10
 
WiFi | Time from NTP | Sat Sep 24 05:31:18 2022
WiFi | signal strength | -35 dBm
WiFi | signal quality | 100 %
 
NAM | Number of measurements: | 3404
NAM | Uptime | 6d 5h 43m 42s
NAM | Time from last update attempt | 5h 32m 12s
 
NAMF | Sensor slots | 10
NAMF | Free slots | 5
NAMF | Sensors | NTW_WTD SHT3x SDS011 HECA BMPx80
 
ESP | Reset Reason | Power On
ESP | Max Free Block Size | 16024/27648 B
ESP | Heap Fragmentation | 2/30 %
ESP | Free Cont Stack | 804/804 B
ESP | Free Memory | 26960/28720 B
ESP | Flash ID | 1458392
ESP | Flash Vendor ID | 216
ESP | Flash Speed | 40.00 MHz
ESP | Flash Mode | 2
 
ENV | Core version | 2_7_4
ENV | SDK version | 2.2.2-dev(38a443e)
 
DEBUG | Debug options |  

</body>
</html>

This is the json content.
`{"software_version": "NAMF-2020-43", "age":"30", "measurements":"3405", "uptime":"539068", "sensordatavalues":[{"value_type":"SDS_P1","value":"6.60"},{"value_type":"SDS_P2","value":"3.40"},{"value_type":"SHT3X_temperature","value":"10.77"},{"value_type":"SHT3X_humidity","value":"53.17"},{"value_type":"HECA_temperature","value":"10.57"},{"value_type":"HECA_humidity","value":"53.44"},{"value_type":"HECA_Tdc","value":"0.00"},{"value_type":"HECA_RHdc","value":"0.00"},{"value_type":"HECA_dc","value":"0.00"},{"value_type":"BMP_pressure","value":"86430.40"},{"value_type":"BMP_temperature","value":"11.50"},{"value_type":"samples","value":"105506"},{"value_type":"min_micro","value":"1278"},{"value_type":"max_micro","value":"99734"},{"value_type":"signal","value":"-34"}]}`