nettigo/namf #61: Network Broken/Device needs regular reboots after update to NAMF-2020-44

jalogisch, 2022-10-20:
Hardware: [KIT 0.3.3](https://nettigo.eu/products/nettigo-air-monitor-kit-bme-0-3-3-pro-soldered-build-your-own-smog-sensor) 
Software: Autoupdated to NAMF-2020-44

Since the auto update from the previous Version my AirMonitor needs regular reboots and it stops reporting and is not reachable via Network. 

I'm not able to debug that device as I do not know what I need to make to get a new (old) firmware on that device when it is not working via Network. Maybe some english instructions how I can debug that situation as I can't get the logs from the device when it is not working. I still see a light flashing when it is not reachable but there is no way for me to get on that device. 

If this is not the right place to look for that kind of help, please let me know where I can go.

xnaldax, 2022-10-27:
I have exactly same problem since last update - KIT BME 0.3.3 PRO. 
@netmaniac

rdbis, 2022-10-28:
Same problem with my NAM KIT BME 0.3.3 PRO. With the latest firmware, device works max 2-3 days from reboot, then it becomes not visible in WiFi network.

netmaniac, 2022-11-26:
Hello,
I'm sorry for delay in response. Regarding issue - can be device pinged when it stops reporting? (on Windows cmd key + r, type cmd, enter, when window opens type in `ping IP_ADDRESS_OF_SENSOR`).

Blue led on Wemos - due to choices made by Luftdaten I2C pins are shared with pin where LED is attached. So blinking LED is related to I2C communication (BME sensor for) not network activity.

I'll prepare info how sensor can be downgraded.

rdbis, 2022-11-26:
Re: can be device pinged when it stops reporting?  - No, device is not responding to ping requests

netmaniac, 2022-11-28:
I have just released new beta software, which allows to rollback version. If You want to use older NAMF version proceed as follows:

- change updated channel to beta (Advanced config tab), save & restart - autopupdate has to be enabled
- wait 30-60 seconds - sensor should boot with NAMF-2020-45rc1
- now visit IP_OF_THE_SENSOR/rollback - You will have option to select older firmware version (currently only EN or PL fw)

After saving sensor will restart and update to older FW version. Autoupdate will disabled. If You want to upgrade to current version just re-enable autoupdate.

Please report if downgrade have fixed that problem with sensor accessibility.

rdbis, 2022-11-28:
Hi, thanks for preparing downgrading feature. I've made an downgrade to v43. So far it looks fine.

jalogisch, 2022-12-15:
I was able to enable the network watchdog - that is making the situation also stable for me.

jalogisch, 2023-12-27:
For a few days, my device is boot-loop again and is never coming up again. 

Not sure how I can fix this properly at the moment as the device is never reachable via network.