nettigo/namf #64: No measurements from BME280 sensor

piotrek-k, 2022-12-18:
I'm having trouble getting BME280 sensor to work. I don't receive any measurements, it also doesn't seem to be detected by microcontroller.

I've got NAM KIT BME 0.3.3 STD. Firmware: NAMF-2020-44

I've done a few checks to see if wiring is set up correctly. What I've checked so far:
* There is 3.3V between VCC and GMD pins while sensor is running
* There is continuity between SCL pin (on I2C port) and D4 pin on Wemos. I've measured resistance using multimeter, which shows approx. 1.1 ohms
* There is continuity between SDA pin (on I2C port) and D3 pin on Wemos. Checked the same way as above

Also:
* sensor in enabled in settings
* measurements from HECA are not shown either (which might indicate that there is a problem with D3/D4 pins?)

How "measurments" page looks like right now:
[image]

Sensor statuses:
[image]

Also, logs from "/debug" page:
[logs.txt](https://github.com/nettigo/namf/files/10253647/logs.txt)

netmaniac, 2022-12-20:
Hello,
If You connect only HECA or only BME there is anything on I2C bus (/status page)?
Also - have You correctly soldered I2C cables for BME/HECA? Are wires in correct order (BME: https://docs.nettigo.pl/en/nam/033/nam-bme280-033 HECA:https://docs.nettigo.pl/en/nam/033/nam-heca-033)

piotrek-k, 2022-12-21:
I've just re-soldiered BME, but still doesn't work. I'm considering testing it separately on a contact plate with another microcontroller...

Connecting only HECA or only BME doesn't change anything on status page. Must be something either with I2C bus or with Wemos itself. Is there an easy way to test whether all Wemos pins are operational?

piotrek-k, 2022-12-27:
Hello,

Small update: I've just mounted all sensors on contact plate and connected them with wires with Wemos and everything seem to be working correctly (I'm able to log in to web server and receive measurements from all sensors). So at this point I'm pretty sure there is a problem with PCB.

I've kept experimenting, re-soldering some parts and at some point web server ceased to start up entirely when connected to PCB. That was happening sometimes previously, but I've ignored it since switching off and on again entire circuit usually solved that issue. Now it doesn't help, but when I launch Wemos separately, using USB port, server correctly starts up and sensors are working. 

**So, to sum up, the current state is:**
* Wemos on PCB main board does not work (cannot connect to web server). It does not work even if I disconnect all sensors and even if I use USB port to power it up.
* Wemos does work and serves webpage after taking it off the board
* There seems to be no shot-circuit between any of pins on Wemos (except GND pin which is connected to any other)
* When PCB board is powered up (by power supply), there is voltage between 5V and GND (in fact, it oscillates between 4 and 6) and between 3.3V pin and GND (oscillates between 3 and 4).