nettigo/namf #67: NTP pool set to Europe

MMBytes, 2023-07-09:
Hi Team, 
can you please not set the NTP pool automatically to a region?
DNS does this automatically.
Please just set it to pool.ntp.org as defined here:
https://support.ntp.org/Servers/NTPPoolServers

I'm in Australia and the NTP server is picked from the Europe list.

Thanks

netmaniac, 2023-09-19:
Hi!
Sorry for delay, in beta 46rc3 will be that fixed, but I would suggest waiting for -46 stable release. In 46rc3 is a lot changes in WiFi so I expect some errors to show :) Do not use beta channel unless You can reinstall Your sensor.

MMBytes, 2024-07-12:
Thanks @netmaniac, any chance of making this a configuration setting in the GUI? I have a local NTP server I want to use please. Thanks

commit 8728a6b0 (2023-09-19): for English language select 0.pool.ntp.org. Should close #67