nettigo/namf #69: SDS011 reply checksum failed and freezing

ggocsei, 2023-10-11:
After a year flawless working, I'm sterted to see SDS011 reply checksum failed in the log. 
I don't know if there any correlation, but lately the device is started to freeze after one-two days. After power off-on it's working as usual (with the checksum failed) and then freeze again in the one-two days period. 
I've no other errors in the log. Send to API Sensor Community , Send diagnostic data to Nettigo settings are enabled. 
The Auto update firmware, using channe also enabled with the stable channel. How can I make it really stable again?

danielskowronski, 2023-12-04:
Same here. However, I was getting this bug randomly over the last couple of years - I got v2 sensor back from times before NAMF fork existed. A few days ago I tried NAMF-2020-45 after using NRZ-2020-133 since it was released, and it hit the same issue. Power supply is not at fault here, the device can draw 300-400mA and not much more because it doesn't negotiate any higher-current charging protocol.

---

From the only log on current stable firmware I have, it seems like after crash, the device does not re-initialize SDS011 and it always errors out with "SDS011 reply checksum failed" until you power it off and on again. However, I didn't find proper timing for that (I'm probably way too impatient with some capacitors or the code is buggy) and the only solution for me was to enter AP mode, re-enter Wi-Fi password and it just started to work. 

The device is connected to the PC to gather debug data shows that it restarts internally, but claims it's unable to get IP so fails to do anything. On the other hand, Wi-Fi router claims device connects to the physical layer and has issued IP address from DHCP.

Here's that log, unfortunately it was on level=3 so not much details:

```
2023-12-04 10:29:27: SDS011: end of cycle
2023-12-04 10:29:33: Creating data string:
2023-12-04 10:29:33: WLAN signal strength: -90 dBm
2023-12-04 10:29:33: ----
2023-12-04 10:29:46: Succeeded - 
2023-12-04 10:29:47: Succeeded - 
2023-12-04 10:29:47: ## Sending to madavi.de: 
2023-12-04 10:29:47: Succeeded - 
{d.lśž|.„lŕ|....Ś.lě.c|‡‚.ě.“s“c„.c„űooźlggśăä.c.x„Źdrlslpűoŕ.....l.Śś...c.nă|.ädŹ.‡cŚűggď.lŚŽl ...gg.d`...o{Ź››o..b.l`..s““n..c.d`.śc..‡..’ls“`.üoś.{d.dśź|.„lŕ|....Ś.lě.c|‡.ä.“{›c„.cŚűogźlggśăä.b.p„Źdslslpűoŕ....‚.l.Śś...b.oă|.äl‡.‡cŚňggď.l„Źd`...nn.d`...o{Ź›Űg..b.l`..s““n..c.l`.ś#..‡..“ls“`.ügś.1970-01-01 00:00:00: 
NAMF ver: NAMF-2020-45/EN
1970-01-01 00:00:00: Chip ID: 1234567
1970-01-01 00:00:00: SPIFFS (kB): 888334
1970-01-01 00:00:00: Free sketch space (kB): 1416
1970-01-01 00:00:00: CPU freq (MHz): 160
1970-01-01 00:00:00: Set defaults
1970-01-01 00:00:00: mounting FS: OK
1970-01-01 00:00:00: SDS011: start
1970-01-01 00:00:00:  [HECA ERROR] Cannot start periodic mode
1970-01-01 00:00:00: Trying BME280 sensor on 
1970-01-01 00:00:00: 76 ... found
1970-01-01 00:00:01: BME280: start
1970-01-01 00:00:01: Config parsed
1970-01-01 00:00:01: PowerOnTest
1970-01-01 00:00:01: Starting Webserver... (IP unset)
1970-01-01 00:00:01: output debug text to displays...
1970-01-01 00:00:01: SSID: 'XXXXXXXXXX'
1970-01-01 00:00:01: 6
1970-01-01 00:00:01: ........................................
1970-01-01 00:00:22: output debug text to displays...
1970-01-01 00:00:22: Failed to connect to WiFi. Trying to connect to fallback WiFi
1970-01-01 00:00:22: XXXXXXXXXX_IoT
1970-01-01 00:00:22: ..........SDS011 reply checksum failed 
1970-01-01 00:00:27: ..............................output debug text to displays...
1970-01-01 00:00:42: Starting WiFiManager
1970-01-01 00:00:42: AP ID: NAM-1234567
1970-01-01 00:00:42: Password: 
1970-01-01 00:00:42: scan for wifi networks...
1970-01-01 00:00:44: Starting AP with default password
1970-01-01 00:01:17: Connecting to XXXXXXXXXX
1970-01-01 00:01:17: ---- Result Webconfig ----
1970-01-01 00:01:17: WLANSSID: XXXXXXXXXX
1970-01-01 00:01:17: ----
Reading ...
1970-01-01 00:01:17: SDS: 0
1970-01-01 00:01:17: PMS: 0
1970-01-01 00:01:17: DHT: 0
1970-01-01 00:01:17: DS18B20: 0
1970-01-01 00:01:17: ----
Send to ...
1970-01-01 00:01:17: Dusti: 1
1970-01-01 00:01:17: Madavi: 1
1970-01-01 00:01:17: CSV: 0
1970-01-01 00:01:17: ----
1970-01-01 00:01:17: Autoupdate: 0
1970-01-01 00:01:17: Display: 0
1970-01-01 00:01:17: LCD 1602: 0
1970-01-01 00:01:17: Debug: 3
1970-01-01 00:01:17: ....SDS011 reply checksum failed 
1970-01-01 00:01:20: ................
[... 565 lines cut ...]
-----> Received 64 Bytes:
2023-12-04 13:37:03 2023-12-04 13:37:03: Starting Webserver... 192.168.XXX.XXX
2023

-----> Received 170 Bytes:
2023-12-04 13:37:04 -12-04 13:37:03: Send to :
2023-12-04 13:37:03: luftdaten.info
2023-12-04 13:37:03: Madavi.de
2023-12-04 13:37:03: 
2023-12-04 13:37:04: Clear factory reset markers

-----> Received 52 Bytes:
2023-12-04 13:37:04 2023-12-04 13:37:04: SDS011 reply checksum failed 

```

---

To sum it up: 

- latest alpha release NAMF-2020-46rc2 seems to make it resilient to Wi-Fi connection issues after SDS011 crash (however, that's only based on short-term observations)
  - this suggests 
- SDS011 is prone to checksum issues - even with longer measurement window
  - why it crashes other parts of the firmware?
  - can this be related to sensor dying from long usage? mine has been running for nearly 5 years now - but the measurements match nearby stations

danielskowronski, 2023-12-04:
It seems like alpha firmware solves freezing of Wi-Fi after crash.

Crashes are 99% hardware related - I discovered a terrible voltage drop on my USB cable (3.6V on 5V test point), USB power supply is quite good. It was probably amplified by temperature around -7 degrees Celsius, so the heater was drawing significant current, dropping voltage further.

Before that change, I had around 24 crashes in less than 10 hours. Will be monitoring overnight and for a few more days.

Maybe we could add monitoring of 5V rail with analogue input A0 and some simple discrete voltage divider or diode? AFAIK ESP8266 should be able to take 3.3V on the built-in ADC, so diagnostic hack should be doable.

netmaniac, 2023-12-12:
> It seems like alpha firmware solves freezing of Wi-Fi after crash.
>
> Crashes are 99% hardware related - I discovered a terrible voltage drop on my USB cable (3.6V on 5V test point), USB power supply [...]

Yes, power supply is crucial. Regarding drop of voltage - if You use USB PS and USB cable to power NAM - then USB cable can be a problem. Heater is PTC, powered by 5V it takes ~700 mA when heater is cold. Current then becomes much lower (100-200 mA AFAIK) when heater has warmed up. So, many cheap USB cables (micro USB especially) have to thin power wires and is huge voltage drop on cable.

Regarding SDS011 checksum failures - they are unavoidable. With software serial on ESP8266 we will always get some timing errors. So failure rate up to 10% is totally OK. 

In other words - checksum errors in log are pretty common and nothing to worry about.

SDS011 and it's lifetime - on standard settings (145 measurement interval, 20 s warmup + measure time) it should last 6-7 years.

danielskowronski, 2024-01-29:
TBH, I couldn't find so cheap USB power supply in my collection - worst got 5V/1A and delivered it. I've personally touched just one so terrible that it didn't conform with USB standards, but it was more of fire hazard ;) 

After some checks, I found out that the cable I used originally delivered 3.7V (1.4V drop), better one 4.2V and best ones I could find (1.2m GreenCell KABGC20, 2m Baseus CAMKLF-CG1) - 4.3V-4.4V (0.7V-0.8V drop). However, it turned out that the reason for my Wi-Fi instability was bugged ASUS-WRT Merlin firmware...

Out of curiosity, I did some checks to find cut-off voltage with a regulated power supply set to constant voltage and voltage measured at screw terminals. It's relatively warm now, and I couldn't find space in my freezer, so measured at ambient 22 C with PTC heating to 35-40 C. All the time current was 0.3A. 

- below 2.4V it does not boot at all (or at least, Wi-Fi radio is silent)
- between 2.4V and 2.7V it boots, enters AP mode and seems to accept settings, but after reboot does not work
- between 2.73V and 3.2V it connects, but SDS011 readings are absurdly high (as expected)
- between 3.3V and 4.0V all works fine, values match other sensor I have, but checksum rate is between 6 and 11% over 5+1 measurements
- above 4.0V (tested up to 5.1V) values again are fine, checksum rate drops to 3-7% over 5+1 measurements

I hope to publish more details soon, but it seems like you'd need to have a very shitty charger (most modern ones support 2.4A) or broken shitty cable (with voltage drop between 1.2V on lowest end of semi-acceptable USB voltage and 1.8V on higher end of range). I can see a scenario when the charger struggles with 0.5A and cable used is of very poor quality, but it seems like it should yield much more spectacular results - unable to connect at all, absurd results or much higher checksum error rates. Plus, it's 2024, I don't believe you wouldn't throw away such a charger.

As for Wi-Fi connection issues, I have no clue what could be the reason as I tried multiple options on both ends, and it only went away after roll-back to stock Asus firmware, but keeping all the settings the same. Now it connects to network after few seconds and gets reasonable ping (3-20ms, some much longer but usually when device exports data), before change it was taking forever to connect, most of the time falling back to AP mode and when it connected it got DHCP lease but wasn't reachable at all. Nothing physical changed - same antennas, same cables, same window and position. Wi-Fi seems to work on so low voltage, you should be able to diagnose power issues that way.