nettigo/namf #72: FR: Add WiFi signal level in web interface

lemielek, 2023-12-06:
If it is possible, please add WiFi signal level in web interface.
Eventually also in data.json if it is not already there? Cannot check now, because of this lost signal. Will probably have to add additional AP outside to have stable connection. I will try also deinstall LCD, maybe then will be better signal.
My NAM-3149890 periodically loses WiFi signal, even when it shows that is on 28% percent level. It is standard Wemos with internal small antenna and soldered in because I did not found new docs when I was soldering it after two years in storage.

netmaniac, 2023-12-12:
On /status page there is both % value and dBm. More data than in `data.json` is sent to Influx DB. It has also WiFi signal level in dBm and some memory usage data. I should probably add those to JSON too.

lemielek, 2023-12-12:
I was blind. Sorry. And thank you both.
Then later when I was configuring it I saw beautiful two lines with networks available to connect.

closed 2024-03-22 by netmaniac