nettigo/namf #76: DS18B20

repcio, 2024-05-08:
I have bought sensors DS18B20 sensors from nettigo and connected them to connector on plate.
Even after change in configuration that module is enabled and connected there are no readings from the sensor.
I have connected all 3 sensors that I have bought but the results is the same.

netmaniac, 2024-05-15:
Have You added 4k7 resistor between DS18b20 Vcc and signal line? It is required by DS and it is not provided on board...

danielskowronski, 2024-05-29:
Same issue, solved by adding a pull-up resistor. 

I found a lazy, but least invasive way I could think of - two "goldpin" F-F cables connected to "EXT" connector to pins that match 3V3 and D7, on other ends of cables - just shove in a bare resistor :)

The trickiest bit in investigating why it didn't work was the fact that DS18B20 is read just before API uploads, while the frequency of uploads is controlled by a setting labelled as "Measuring interval".

netmaniac, 2024-08-20:
In upcoming 47rc1 information about need resistor will be shown on config page (using new [help system](https://blog.nettigo.eu/2024/namf-2020-46rc7/)).
In next PCB revision I'll try to add place for this resistor.

closed 2024-08-20 by netmaniac