nettigo/namf #77: Issues on ESP82668266 with 16MB flash

danielskowronski, 2024-05-29:
## Summary

When you use ESP8266 that has 16MB flash instead of 4MB, system performance is non-existent, most notably config page does not load 99% of time.

## Tests

Scenario:

1. Flash fresh ESP8266 that has 16MB with any version of NAMF (I believe the same may be the case for original firmware). I was doing most of my tests on `NAMF-2020-46rc5-en.bin`
2. Connect to the board AP mode with any computer that's close to the board. In my environment, NAMF was reporting stable -31 dBm signal to client device.
3. Perform `time curl -v http://192.168.4.1/config -o /dev/null` several times

Expected outcome:

- config page on average is fetched nearly instantly (less than 0.5 seconds)

## Background

ESP8266 usually has 4MB SPI flash. However, some boards marketed as "WeMos D1 Mini Pro" that are preferred for their external antenna are sold with 16MB SPI flash.

curl: (56) Malformed encoding found in chunked-encoding

netmaniac, 2024-05-30:
Thank You for effort you have put into this. Now only few words on topic, I'll try to write a bit more in few days.

Issue 7095 You have mentioned - I believe they say about 16M filesystem not chip size. AFAIK SPIFFS ignores all not used flash, so from data structure filesystem has 2M. I have observed that behavior, and it is one of the reasons we don't sell Wemoses D1 mini Pro. Apparently boards from some manufactures are OK and are working good, and some have this problem. So I'm not sure if this is a software or hardware problem, but I can't rule out either cause. Do Your Wemos have antenna/IPX selecting jumper placed under angle 45° to ceramic antenna? 

Regarding changing design of how webconfig works - I did a test with such approach, but since current ESP8266WebSever is single threaded and blocking - performance (as seen by user) was dramatic. I did test with replacing it with [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer?tab=readme-ov-file) there was much better user experience, but I had some other problems with this server. Unfortunately I can not find my notes from this project so I can not say what was exactly problem. I believe that [async_web](https://github.com/nettigo/namf/tree/async_web) is this code.

danielskowronski, 2024-06-11:
No, my "pro" board has an antenna resistor at 90° (it's that weird rare variant). 

After running into some issues with the standard 4MB Wemos D1 Mini (the one sold as part of NAM kit), I'm not so sure if it's SPIFFS that causes all the issues (but probably still makes it harder to debug). 

My NAM 0.3.3 box was working for weeks, but today it disconnected from Wi-Fi and refused to reconnect in a stable manner (when it connected, every endpoint was tragically slow). I actually investigated the issue live for the first time, but it wasn't the first occurrence (although it always fixed itself). 

After desperate attempts at re-flashing, I even faced full firmware crashes, but they were probably caused by corrupted config.json stored on SPIFFS. I thought it could be related to https://github.com/esp8266/Arduino/issues/6007 or https://github.com/esp8266/Arduino/issues/5493, but my ancient Wemos from NAM 0.2.1 exhibits the same issue with wake-ups; however, it never actually failed on me with Luftdaten/NAMF. Validated flash by dumping and it's working fine. I even tried lowering Wi-Fi TX power to try keeping flash stable.

I've seen similar issues with SPIFFS + multipart HTTP + ESP8266 in multiple places, but nothing really stood out. I even tried building an image with even more debug and random changes to `webserverPartialSend` trying to correlate content sent with content received. That, plus going one layer deeper, was the way! Since `curl` was vague in telling me what actually broke in the HTTP response, I looked at network traffic in Wireshark. And I've seen a sea of red and black lines for spurious retransmits and other out-of-order TCP packets. 

After some tweaking, I reset TX power to max (20.5, even if code accepts any number) and changed Wi-Fi mode to 802.11**b**. I never had luck with N when connecting to my Wi-Fi router, so long time ago, I assumed it had to do with some compatibility issue and stuck to G. However, with B mode, the connection is the most stable and retransmits are very rare. My current theory is that the ESP8266 running NAMF must be somehow so overloaded, it struggles to run more modern variants. 

The most bizarre twist: that 16MB "pro" board started to work with 802.11**b** - it's relatively slow, but renders config page nearly 100% of the time. 

With 802.11b and 20.5dBm TX power on rc5, the config page renders on those boards in:

- 4MB (shielded; SPI Flash `0x5E` - Zbit) from nettigo.pl in assembled kit 0.3.3, placed outside, -76dBm signal and one more hop from my machine (wired AP) -> 0.5-1.0s
- 16MB "pro" (ESP8266 chip directly on board; SPI Flash `0x5E` - Zbit) in empty breadboard, placed 2-3 meters from main router, -45dBm signal -> 1.25-1.75s
- 4MB (most likely the only genuine Wemos here - shielded ESP8266MOD board soldered to Wemos board; SPI Flash `0xEF` - Winbond) from nettigo.pl in assembled kit 0.2.1, placed 2-3 meters from main router, -46dbm signal -> 0.2-0.3s

That suggests off-brand SPI flash is significantly slower, at least with SPIFFS.

@netmaniac I've seen some work done on ESP32 variant - how it's going? Ideally, we'd need new main board to support it, but I think that using wire spaghetti to convert some of the available dev boards with ESP32 would be worth trying to fix performance issues in the vast ocean of non-genuine D1 Mini variants. ESP32-CAM seems the easiest board to adapt due to low price, similar size and built-in external antenna connector (we could just ignore dangling camera and take advantage of microSD for logs storage).

danielskowronski, 2024-06-13:
I am beginning to wonder if off-brand SPI is just an indicator of poor quality of ESP8266 dev board, rather than root cause. 

---

After some tweaks to my Asus Wi-Fi router (I lost track of what's causing most of the issues, but most likely *Airtime Fairness* has to be turned off) and sticking to `WIFI_PHY_MODE_11B` I've run some tests, and it starts to be more of a network issue. I was hinted towards that direction by discrepancies in `/config` vs `/config.json` pages - their size is different and chunked response generates many TCP packets, which likely generates some issue similar to memleak, but on TCP stack.

I developed a "test suite" that runs curl to `/config` 1000 times and records curl exit code (0 for OK, 56 for malformed chunked response) plus execution time. That way, results are somehow comparable and if any memory leak is present it should be visible by series of non-stop requests. Those are executed from a machine connected using cable to router, against firmware run on 16MB "pro" board that started the issue running in forced B mode. Everything compiled using this new platformio target: https://github.com/danielskowronski/namf/commit/6510b4e950b5a4b027117a88cd2c323466ce40be#diff-4446afd728a4f34cbcddc306a9cb6be845d1a61c216076a295683bcc9c106724

One thing that should be ruled out is my Wi-Fi setup, ideally to get reproducible results, it should be re-run on some Raspberry Pi.

First thing I checked was `server.client().setNoDelay(1);` set before each chunk in `webserverPartialSend` (see https://github.com/nettigo/namf/blob/beta/src/webserver.cpp#L11):

- `setNoDelay(1)` (from upstream) - 1.2% errors, 1.49442s average response time; lots of spurious retransmission
- `setNoDelay(0)` - 1.1% errors, 0.7585s average response time

And this seems to make sense after reading https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/client-class.html#setnodelay and the linked Wikipedia article about delayed ACK. Especially when using slow 802.11b/g and having ESP8266 outdoor (plus probably slow flash doesn't help).

This is an example of transmission with Naggle enabled (`setNoDelay(1)`) - it's working against delayed ACK as chunked HTML is further chunked into smaller TCP packets:

[image]

And this is with Naggle disabled - chunking only happens to fit into Ethernet frames:

[image]

Again, it seems that if anything goes wrong with a single TCP packet, poor ESP8266 has to retransmit it, deal with other packets that are now out of order and ultimately fails to render something that makes sense in a reasonable time, thus forcing the client to just terminate connection. 

I'll post full test results for WiFi mode x Naggle x board (4MB & 16MB from 2024 plus 4MB from 2019 that's genuine Wemos with 99% certainty). Although from partial tests with 16MB board that switched from B to G, I already started to see original behaviour with extremely long response times.

danielskowronski, 2024-06-13:
Mystery of N-mode not working is solved - AP mode is known to work only in B/G, STA should work, but Asus (and some other vendors) are stubborn in following 802.11n specs and silently ignore connections from certain SDK versions that do not advertise WMM (it manifests as established link-layer connection, but no connectivity - most notably no DHCP). Best described on https://github.com/esp8266/Arduino/issues/7965

Moreover, the linked issue also mentions issues with poor quality crystal oscillator that also may be influencing connectivity, especially with TCP. I constantly observe random disconnects on all new (aka non-Wemos) boards when used outside. If that's confirmed, then I think there's no future for supporting ESP8266 if we can't get good quality boards (and we can't get Wemos ones, since they no longer make them).

For me, that was the last nail in the AsusWRT coffin, so I'll soon have Mikrotik setup to play around and determine the true root-cause of those connectivity issues on the NAM end. If nothing else, I'll finally get rid of the incoherent mess that's Wi-Fi configuration on AsusWRT...

But for now, I added `PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190313` and it's still not connecting to as N client (I suspect Asus made further "adjustments" to their firmware). On G mode there are huge packet losses, so only B mode works, but that one struggles outdoor...