nettigo/namf #78: [QUESTION] `data.json` with GPS data

bieniu, 2024-06-01:
I would like to ask for an example of `data.json` from a device with GPS connected. I would like to implement GPS data support in https://github.com/bieniu/nettigo-air-monitor

netmaniac, 2024-06-29:
Hello,
I want to apologize for lack of reply. I was focused on other topics recently and I have missed notification from Github. Here is `data.json` with enabled and working GPS.
```
{
  "software_version": "NAMF-2020-46rc7",
  "age": "7",
  "measurements": "1",
  "uptime": "49",
  "sensordatavalues": [
    {
      "value_type": "GPS_lat",
      "value": "52.284921"
    },
    {
      "value_type": "GPS_lon",
      "value": "20.889263"
    },
    {
      "value_type": "GPS_height",
      "value": "102.80"
    },
    {
      "value_type": "GPS_date",
      "value": "06/29/2024"
    },
    {
      "value_type": "GPS_time",
      "value": "12:18:39.00"
    },
    {
      "value_type": "samples",
      "value": "25371"
    },
    {
      "value_type": "min_micro",
      "value": "289"
    },
    {
      "value_type": "max_micro",
      "value": "35392"
    },
    {
      "value_type": "signal",
      "value": "-48"
    }
  ]
}
```

bieniu, 2024-06-30:
No worries, thanks!

closed 2024-06-30 by bieniu