nettigo/namf #79: Metrics with Prometheus

wal-and, 2024-11-18:
Below is the result of the metrics query:

`software_version{version="NAMF-2020-46",node=")esp8266-XXXXXXXX"} 1
uptime_ms{node=")esp8266-XXXXXXXX"} 9349402
sending_intervall_ms{node=")esp8266-XXXXXXXX"} 145000
number_of_measurements{node=")esp8266-XXXXXXXX"} 64
SDS_P1{node=")esp8266-XXXXXXXX"} 0.50
SDS_P2{node=")esp8266-XXXXXXXX"} 0.20
HECA_temperature{node=")esp8266-XXXXXXXX"} 12.10
HECA_humidity{node=")esp8266-XXXXXXXX"} 51.88
HECA_Tdc{node=")esp8266-XXXXXXXX"} 0.00
HECA_RHdc{node=")esp8266-XXXXXXXX"} 0.00
HECA_dc{node=")esp8266-XXXXXXXX"} 0.00
BME280_temperature{node=")esp8266-XXXXXXXX"} 12.55
BME280_pressure{node=")esp8266-XXXXXXXX"} 99613.66
BME280_humidity{node=")esp8266-XXXXXXXX"} 46.65
samples{node=")esp8266-XXXXXXXX"} 99327
min_micro{node=")esp8266-XXXXXXXX"} 307
max_micro{node=")esp8266-XXXXXXXX"} 180528
last_sample_age_ms{node=")esp8266-XXXXXXXX"} 67262`

Why, each node name starts with a closing bracket?

netmaniac, 2024-11-21:
This was mistake introduced when reformatting code. Since I don't use that data format and nobody reported before thus was unnoticed.

I will make new stable release with this fix, so in few days You should have correct data.

closed 2024-11-21 by netmaniac

netmaniac, 2024-11-23:
There is a new binary on update server - NAMF-2020-46a with this error fixed.

commit 574be23b (2024-11-21): removing ) placed by mistake in prometheus format. Should fix #79