nettigo/namf #82: Brak wykrywanie BME280 na szynie i2c D1 mini pro

Jabol86, 2025-02-16:
Witam.

Buduje czujnik smogu bez waszej płytki, podłączając sie bezpośrednio pod płytke d1 mini pro.

Niestety podłączając termometr pod porty D3 i D4 nie jest wykrywany termometr. Sprawdzałem czy jest sprawny i pod esphome na portach d3 i d4 również nie ma detekcji dopiero na porcie tx i rx.

Natomiast na płytce zwykłej d1 mini wszystko działa.

Spotkał sie ktoś z takim problemem?

closed 2025-02-17 by Jabol86

netmaniac, 2025-02-24:
Udało się ustalić co było przyczyną?

Jabol86, 2025-02-24:
> Udało się ustalić co było przyczyną?

Tak. Okazała się trefna seria płytek od kolegi chińczyka z wemos d1 w wersji pro.

Jabol86, 2025-02-27:
> Tak, niestety teraz jest pełno trefnych Wemosów Pro dlatego zrezygnowaliśmy z nich na rzecz Bidamosa. Albo I2C nie działa, albo cuda z zasięgiem i stabilnością WiFi.

Rozumiem :(

Nie planujecie może dodać opcji obsługi:
-  WS2811 jako sygnalizacje jakości powietrza?
- obsługę czujnika NOX
- możliwość wyboru pinów dla i2c bądz obsługi sht31 pod adresem 44 jeśli nie mamy grzałki ? (na rynku dużo sht bez możliwości zmiany adresu)